yaacov/node-modbus-serial #500: Connection timesout when connecting to multiple clients to modbus server in parallel over modbus tcpip connection

maitreyi-timesys, 2023-04-20:
I am using modbus-serial package to connect to EthernetPLC using TCPConnect() method.
My program continuously reads and writes registers from PLC.
I have 3 Clients in parallel which will connect to Ethernet PLC device.
But when I connect 1, the other one gives connection timeout Error.
when other one connects then Port not open error on previous Client is noticed.

Interestingly, when I connect my program with modbus-SIMULATOR exe, (https://sourceforge.net/projects/modrssim/) I can seemlessly connect multiple devices and run the same application, without getting any communication error.

indroneelray, 2023-04-29:
I have this issue too. I've been currently extracting data from 1-2 PLCs for the demo.  I'm in a situation where my end product will be reading data from approximately 800 PLCs. I am extremely worried that when we get to that point, this method of extracting data might not hold up. I'm eagerly waiting for a response from the author of this repo

indroneelray, 2023-08-20:
@maitreyi-timesys did you manage to figure this out?

maitreyi-timesys, 2023-08-24:
No solution as yet.