yaacov/node-modbus-serial #505: ModbusRTU returns  ETIMEDOUT even for invalid addresses

volkmarnissen, 2023-07-23:
This is how to **reproduce the issue**:

Connact a device to USB RS485 port on raspberry pi
read holding register of an address which is not available on the device.

Instead of a modbusErrorCode , you will get "ETIMEDOUT" exceptions with no detailed information about the reason.
The device, I am using seems to have 14 valid addresses. Addresses higher than 14 will return with ETIMEDOUT.
I was expecting them to return modbusErrorCode 0x02 (Illegal Address).

The device itself seems to have an issue. Every second call to the device fails. Other devices work more reliably.
However, I suspect an modbusErrorCode might help here also to analyze the issue.

**Questions:**
Is this the expected behaviour?
I am not familiar with the code and with debugging it. How can I enable debug logging?
Can you give me a hint where I can start analyzing the issue?
Do you have a development guide?

This is my **test code**:

```
const ModbusRTU = require("modbus-serial");
console.log("New");
const client = new ModbusRTU();
let k = 0;
let addr = -1;
let retries = 0;

// open connection to a serial port
console.log("Connect");
client.connectRTUBuffered("/dev/ttyACM0", { baudRate: 9600 }).then(() => {
    console.log("next");
    client.setID(1);
    client.setTimeout(100);
    read({ data: "start" });
});
console.log("Connect done");

function read(value) {
    k++;
    if (value.data !== "retry") {
        retries = 0;
        addr++;
    }
    console.debug("Modbus readHoldingRegister returned: " + addr + "\n" + JSON.stringify(value));
    if (client.isOpen)
        console.debug("isOpen");
    if (k < 1025) {
        client.readHoldingRegisters(addr, 1).then(read).catch((e) => {

            console.log("ERROR: " + addr + "\n" + JSON.stringify(e));
            if (retries++ < 10)
                read({ data: "retry" });
        });
    } else
        client.close(() => { });
}
```
This is the response
```
New
Connect
Connect done
next
Modbus readHoldingRegister returned: 0
{"data":"start"}
isOpen
Modbus readHoldingRegister returned: 1
{"data":[1],"buffer":{"type":"Buffer","data":[0,1]}}
isOpen
ERROR: 1
{"name":"TransactionTimedOutError","message":"Timed out","errno":"ETIMEDOUT"}
Modbus readHoldingRegister returned: 1
{"data":"retry"}
isOpen
Modbus readHoldingRegister returned: 2
{"data":[3],"buffer":{"type":"Buffer","data":[0,3]}}
isOpen
ERROR: 2
{"name":"TransactionTimedOutError","message":"Timed out","errno":"ETIMEDOUT"}
Modbus readHoldingRegister returned: 2
{"data":"retry"}
isOpen
Modbus readHoldingRegister returned: 3
{"data":[1],"buffer":{"type":"Buffer","data":[0,1]}}
isOpen
ERROR: 3
{"name":"TransactionTimedOutError","message":"Timed out","errno":"ETIMEDOUT"}
Modbus readHoldingRegister returned: 3
{"data":"retry"}
isOpen
Modbus readHoldingRegister returned: 4
{"data":[1],"buffer":{"type":"Buffer","data":[0,1]}}
isOpen
ERROR: 4
{"name":"TransactionTimedOutError","message":"Timed out","errno":"ETIMEDOUT"}
Modbus readHoldingRegister returned: 4
{"data":"retry"}
isOpen
Modbus readHoldingRegister returned: 5
{"data":[216],"buffer":{"type":"Buffer","data":[0,216]}}
isOpen
ERROR: 5
{"name":"TransactionTimedOutError","message":"Timed out","errno":"ETIMEDOUT"}
Modbus readHoldingRegister returned: 5
{"data":"retry"}
isOpen
Modbus readHoldingRegister returned: 6
{"data":[0],"buffer":{"type":"Buffer","data":[0,0]}}
isOpen
ERROR: 6
{"name":"TransactionTimedOutError","message":"Timed out","errno":"ETIMEDOUT"}
Modbus readHoldingRegister returned: 6
{"data":"retry"}
isOpen
Modbus readHoldingRegister returned: 7
{"data":[2000],"buffer":{"type":"Buffer","data":[7,208]}}
isOpen
ERROR: 7
{"name":"TransactionTimedOutError","message":"Timed out","errno":"ETIMEDOUT"}
Modbus readHoldingRegister returned: 7
{"data":"retry"}
isOpen
Modbus readHoldingRegister returned: 8
{"data":[7212],"buffer":{"type":"Buffer","data":[28,44]}}
isOpen
ERROR: 8
{"name":"TransactionTimedOutError","message":"Timed out","errno":"ETIMEDOUT"}
Modbus readHoldingRegister returned: 8
{"data":"retry"}
isOpen
Modbus readHoldingRegister returned: 9
{"data":[1000],"buffer":{"type":"Buffer","data":[3,232]}}
isOpen
ERROR: 9
{"name":"TransactionTimedOutError","message":"Timed out","errno":"ETIMEDOUT"}
Modbus readHoldingRegister returned: 9
{"data":"retry"}
isOpen
Modbus readHoldingRegister returned: 10
{"data":[2],"buffer":{"type":"Buffer","data":[0,2]}}
isOpen
ERROR: 10
{"name":"TransactionTimedOutError","message":"Timed out","errno":"ETIMEDOUT"}
Modbus readHoldingRegister returned: 10
{"data":"retry"}
isOpen
Modbus readHoldingRegister returned: 11
{"data":[1734],"buffer":{"type":"Buffer","data":[6,198]}}
isOpen
ERROR: 11
{"name":"TransactionTimedOutError","message":"Timed out","errno":"ETIMEDOUT"}
Modbus readHoldingRegister returned: 11
{"data":"retry"}
isOpen
Modbus readHoldingRegister returned: 12
{"data":[223],"buffer":{"type":"Buffer","data":[0,223]}}
isOpen
ERROR: 12
{"name":"TransactionTimedOutError","message":"Timed out","errno":"ETIMEDOUT"}
Modbus readHoldingRegister returned: 12
{"data":"retry"}
isOpen
Modbus readHoldingRegister returned: 13
{"data":[14247],"buffer":{"type":"Buffer","data":[55,167]}}
isOpen
ERROR: 13
{"name":"TransactionTimedOutError","message":"Timed out","errno":"ETIMEDOUT"}
Modbus readHoldingRegister returned: 13
{"data":"retry"}
isOpen
Modbus readHoldingRegister returned: 14
{"data":[20739],"buffer":{"type":"Buffer","data":[81,3]}}
isOpen
ERROR: 14
{"name":"TransactionTimedOutError","message":"Timed out","errno":"ETIMEDOUT"}
Modbus readHoldingRegister returned: 14
{"data":"retry"}
isOpen
ERROR: 14
{"modbusCode":2}
Modbus readHoldingRegister returned: 14
{"data":"retry"}
isOpen
ERROR: 14
{"name":"TransactionTimedOutError","message":"Timed out","errno":"ETIMEDOUT"}
Modbus readHoldingRegister returned: 14
{"data":"retry"}
isOpen
ERROR: 14
{"modbusCode":2}
Modbus readHoldingRegister returned: 14
{"data":"retry"}
isOpen
ERROR: 14
```

yaacov, 2023-08-29:
should be fixed by https://www.npmjs.com/package/modbus-serial/v/8.0.13

sigma-design, 2023-09-19:
It seems that every second request in your example fails because in the 'next' step the address is incremented by 1 while 32-but Floating Point values occupy 2 Modbus RTU registers, so the address needs to be incremented by 2.
The response to incorrect addresses depends on Slave unit implementation (not related to Master) - in your case it seems that the slave unit simply ignores requests to invalid addresses (whether it's incorrect offset or non-existing registers), therefore the master produces timeout error eventually.

modbus2mqtt, 2024-02-21:
Thank you,
Sorry, I was busy with other things. Now I tested it: I works perfectly