yaacov/node-modbus-serial #506: Swapped float point

giegiey, 2023-07-27:
Hello, I tried to convert the data result to swapped float point but it did not work .

[image]

This is the result using modbus32.

How to get data result in swapped float point. Please advise this

aaaatttyy, 2023-09-16:
Hi any updates on this I have a similar use case ?

yaacov, 2023-09-16:
try not to swapBytes yourself, find a buffer to float method, that swap it for you:
https://www.geeksforgeeks.org/node-js-buffer-readfloatbe-method/

aaaatttyy, 2023-09-17:
> try not to swapBytes yourself, find a buffer to float method, that swap it for you: https://www.geeksforgeeks.org/node-js-buffer-readfloatbe-method/

Sorry not able to get your point, my usecase is
`We get a buffer from readInputRegisters but since smartEnergyMeter specifies only flaot values to be used as also depicted in the screenshot above I need you recommendation on how to proceed to get the desired output like 240.234, by utilising the buffered returned by the readInputRegister API.`

yaacov, 2023-09-17:
here is an example:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/buffer.js

you need to remove the `swapBytes` method and use one of nodejs built in methods to do the swapping for you.

a. test that the raw data you get is the same the one in modscan, e.g. hex in modscan, and see that the hex value are the same for the same address.
b. different ways to read the data, did you try use big-endian instead of little? https://nodejs.org/api/buffer.html#bufreadfloatbeoffset

aaaatttyy, 2023-09-17:
> https://github.com/yaacov/node-modbus-serial/blob/master/examples/buffer.js

yes I tried these methods already but all of these return a 32 bit int value whereas I wanted it to be something of sort decimal like `240.324` etc.
For point a ,I tested in modscan the hex values are exactly the same,hence quite confused with this one.

yaacov, 2023-09-17:
have you tried `readFloatBE` ?

aaaatttyy, 2023-09-18:
{
  data: {
    data: [
      43909,     0,     0,
          0,     0, 17059,
      51118, 17269, 43909,
          0
    ],
    buffer: <Buffer ab 85 00 00 00 00 00 00 00 00 42 a3 c7 ae 43 7
5 ab 85 00 00>
  }
}

yaacov, 2023-09-18:
I'm adding a help wanted because your problem sound very weird

this is how 240.324 looks fp and swapped fp:
hex: 4370 52f2
fp:                    buf.readFloatBE(0) = 240.32400512695312
swapped fp:   buf.readFloatLE(0) = -4.168164361781354e+30

in your example I can't find a 2 registrar (32 bit) hex that can represent a 0 to 1000 number using swapped fp (readFloatLE) 

assuming this is the same hex input used by modscan, I'm adding a help wanted in case someone knows what the issue is

yaacov, 2023-09-18:
```js
const hexString = 'ab85000000000000000042a3c7ae4375ab850000';
const buf = Buffer.from(hexString, 'hex');

for (let i = 0; i < buf.length - 3; i++) {
    const floatValue = buf.readFloatLE(i);
    if (floatValue >= 0 && floatValue <= 1000) {
        console.log(`Found float value ${floatValue} at position ${i}`);
    }
}
```

hmm, i can actually find some numbers:

```
Found float value 4.795103215073092e-41 at position 0
Found float value 1.8637269575520067e-43 at position 1
Found float value 0 at position 2
Found float value 0 at position 3
Found float value 0 at position 4
Found float value 0 at position 5
Found float value 0 at position 6
Found float value 32 at position 7
Found float value 349.5596618652344 at position 11
Found float value 1.227562818250744e-38 at position 15
Found float value 4.795103215073092e-41 at position 16
```

yaacov, 2023-09-18:
so using your input in position 11 we have somthing that looks like usefull data ...

sigma-design, 2023-09-18:
Historically Modbus RTU was designed only for integer value type using 2-byte 'registers', so there was no definite specification for 4-byte IEEE-754 floating point values. As the result, there are 4 different ways to interpret/convert bytes buffer into IEEE-754 floating point values, see for example settings in the Modbus Poll tool:
[image]

Also Modbus RTU uses "Register numbering" starting from 1 while the actual addressing starts from 0, e.g. Register 30001 means that Modbus RTU reading address should be set to 0 (3xxxx means using "Read Input Registers" command) - this also may lead to incorrect data reading/conversion.

There is a very handy conversion tool [https://www.h-schmidt.net/FloatConverter/IEEE754.html](url) that can be used to try different combinations.

For some reason your example buffer is not multiple of 4x - either there is an additional one byte or missing three bytes.

sigma-design, 2023-09-18:
I am using this library for a while reading floating points values from our instruments, so please consider the following that may be helpful in your case:
1. Make sure that your data buffer for floats has an appropriate number bytes (multiple of 4).
2. Make sure that a correct starting address is used.
3. Try different 32 Bit Float conversion formats since it's not stipulated by Modbus RTU standard and companies can implement it as they see fit for their business.