yaacov/node-modbus-serial #528: An abstraction layer over the modbus protocol

yarosdev, 2023-11-24:
Hello, I have created a great wrapper over modbus, yaacov/node-modbus-serial are the only one driver for now:)

https://github.com/yarosdev/modbusdb

If you dont mind, pls check it, and could you place a link in your readme file?

ps. it is not an issue ) I believe it will helps a lot

yarosdev, 2023-11-24:
worker directory can be removed from your repo, which I have implemented long time ago

yaacov, 2023-11-25:
hi, thank you for the issue and the wrapper package :tada: 

a. you can make a pull request adding an example in the examples directory
b. you can make a pull request adding a new section under the examples section that talk about packages using modbas-serial with the new wrapper and a link to the new wrapper repository.