yaacov/node-modbus-serial #529: Port Not Open

vinit21, 2023-11-26:
TCP Logger  - 

I am pooling 10 modbus devices every 3 seconds. It works well for first 3 minutes and then end up with port not open error. Please find the screen shot of the time where it stopped working. 

When it is working - it is just a beautiful piece of software. I am super grateful to the team for keeping it so updated.  Pooling is faster than kepware. Is there any way we can donate to the team. 

[image]

yaacov, 2023-11-26:
hi, thank you for the issue

adding a help wanted label, in case someone is having the same issue and can help.

look at the examples directory, here is an example that handle errors that should recover from errors:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/logger_complete.js

vinit21, 2023-11-27:
I found the error. Lib is perfectly working. 
It was a silly mistake from my end. Sorry for the trouble.

closed 2023-11-27 by yaacov

IlCallo, 2025-05-07:
@vinit21 can you share which was your mistake?
Cause I'm getting the same behavior and I may be doing the same one