yaacov/node-modbus-serial #530: Error after updating to 8.0.15

zN3utr4l, 2023-11-27:
What should I pass in these 2 fields? Up until version 8.0.13 I never passed anything to these fields

Type '{ port: number; }' is missing the following properties from type 'TcpPortOptions': socket, socketOptsts

```
const options: TcpPortOptions = { port: 502 }

this.client = new ModbusRTU()
    
this.client.connectTCP(ip, options)

// Error
    return new TSError(diagnosticText, diagnosticCodes, diagnostics);
           ^
TSError: ⨯ Unable to compile TypeScript:
services/modbus-service.ts:58:11 - error TS2739: Type '{ port: number; }' is missing the following properties from type 'TcpPortOptions': socket, socketOpts

58     const options: TcpPortOptions = {
             ~~~~~~~

    at createTSError (D:\Repository\test\server\node_modules\ts-node\src\index.ts:859:12)
    at reportTSError (D:\Repository\test\server\node_modules\ts-node\src\index.ts:863:19)
    at getOutput (D:\Repository\test\server\node_modules\ts-node\src\index.ts:1077:36)
    at Object.compile (D:\Repository\test\server\node_modules\ts-node\src\index.ts:1433:41)
    at Module.m._compile (D:\Repository\test\server\node_modules\ts-node\src\index.ts:1617:30)
    at Module._extensions..js (node:internal/modules/cjs/loader:1310:10)
    at Object.require.extensions.<computed> [as .ts] (D:\Repository\test\server\node_modules\ts-node\src\index.ts:1621:12)
    at Module.load (node:internal/modules/cjs/loader:1119:32)
    at Function.Module._load (node:internal/modules/cjs/loader:960:12)
    at Module.require (node:internal/modules/cjs/loader:1143:19) {
  diagnosticCodes: [ 2739 ]
}
```

yaacov, 2023-11-27:
Hi, thank you for the issue :+1:

can you try the fix in #531 ?

closed 2023-11-27 by yaacov

yaacov, 2023-11-27:
https://www.npmjs.com/package/modbus-serial/v/8.0.16 - :tada: :cake: 

@zN3utr4l 8.9.16 should have the fix, can you check ?

zN3utr4l, 2023-11-27:
Work