yaacov/node-modbus-serial #534: Usage of isDebugEnabled to check the PDU raw data

Liviu-Ionel-Anton, 2023-12-05:
How can I use isDebugEnabled to check the raw data? I see that isDebugEnabled is not exposed by the ModbusRTU class.

yaacov, 2023-12-05:
Thank you for the issue.

`isDebugEnabled` is defined here:
https://github.com/yaacov/node-modbus-serial/blob/master/index.js#L623-L630

an example of use, can be found here:
https://github.com/yaacov/node-modbus-serial/blob/master/test/test.js#L301-L315

`request` is defined when using `isDebugEnabled`

can you make a pull request adding an example using `isDebugEnabled` ?
see other examples in:  https://github.com/yaacov/node-modbus-serial/tree/master/examples

i am also adding a "help needed" in case some one have exprience using `isDebugEnabled` and can help with use cases.

Liviu-Ionel-Anton, 2023-12-06:
Hi!

I'm using node.js v18.17.1 and typescript for the project.
I have something like this:

```
import ModbusRTU from "modbus-serial";

tcpPortOptions = {
  ip: "127.0.0.1",
  port: 502,
  timeout: 500
};

const device: ModbusRTU = new ModbusRTU()
await device.connectTCP("127.0.0.1", tcpPortOptions);

device.isDebugEnabled = true;
```

I get : error TS2339: Property 'isDebugEnabled' does not exist on type 'ModbusRTU'.

When I checked the definition of ModbusRTU it leads to ModbusRTU.d.ts file where isDebugEnabled is not defined.

Any help will be appreciated.

yaacov, 2023-12-06:
@Liviu-Ionel-Anton hi, thank you for the followup, did you try addingthe pProperty 'isDebugEnabled' on type 'ModbusRTU  to ModbusRTU.d.ts ?

did it help ?
if it did, can you make a pull request with the fix ?