yaacov/node-modbus-serial #539: Open-close function

tjarb, 2023-12-21:
Hi All,

I try to implement the TCP method this for a Homey device with multiple SUB-devices. Because of that I need the connection to be closed after data exchange. 
Is there a way to open and close the client from within a interval function?

Best regards

yaacov, 2023-12-21:
hi, thank you for the issue, I added help wanted label in case someone knows an answer to this quiestion

yaacov, 2023-12-21:
p.s.
You can look at the examples:
https://github.com/yaacov/node-modbus-serial/blob/master/examples

I think this one may be helpful for your question: 
https://github.com/yaacov/node-modbus-serial/blob/master/examples/polling_TCP.js

try to close the connection after each data collection, wait for the connection to close and then re connect.

tjarb, 2023-12-21:
Hi @yaacov ,

I now use the following:
``` 
const client = new ModbusRTU();
let self = this;

self.pollingInterval_fast = self.homey.setInterval(() => {  
     client.connectTCP(self.getSetting('address'), { port: 502 }).then((result)=>{	

                client.setID(1);
		client.setTimeout( self.getSetting('polling') * 1000 );
																	
		client.readHoldingRegisters(10, 1, function(err, data) {
				if(data.data) self.log("10 " + data.data);
				else self.log(err);
		});
     });	
   /* >>>client seems to have connection closed here <<< */
}, self.getSetting('polling') * 1000);
```

Perhaps I should add the client.close, but when testing the client for connection it's already 'disconnected'.

Best regards,

tjarb, 2023-12-21:
Hi @yaacov ,

it's very strange. If I use the:
```
while(){
   let  ModbusRTU = require("modbus-serial");
   let client = new ModbusRTU();
   client.setTimeout( 1000 );
   client.connectTCP(){
      client.readHoldingRegisters()
   }

   client.close();
}
```
Then there never happens, no data. When I remove the client.close it opens (and keeps open) connections until the modbus server rans out of connections (always 16 sessions).

yaacov, 2023-12-21:
did you try closing the connection after using the read data ?
`client.readHoldingRegisters( ... ).then( ... close() )`

tjarb, 2023-12-21:
Hi @yaacov ,

Here the snipped:

```
const modbus = require('jsmodbus');

class DucoMasterModbusDevice extends Device {
	/*constructor?*/
    async onInit() {
        let self = this;
		
		let i=0;
		const  ModbusRTU = require("modbus-serial");
		
		self.pollingInterval_fast = self.homey.setInterval(() => {  
					self.log("####### "+ i++ +" poll to connect socket  ################");			

					let client = new ModbusRTU();		

					client.setTimeout( 1000 );
					
					client.connectTCP("192.168.2.137", { port: 502 })
						.then( function(){					
							client.setTimeout( 1000 );
							client.setID(1);				
							
							client.readHoldingRegisters(10, 1)
							.then( function(data){
								 self.log("10 " + data.data);			
							})
							.catch(function(e){		
								self.log(e);
							});						
						
					})	
					.catch(function(e){		
						self.log(e);
					});						
				
					client.close();
					
		}, 2000);	//end of pollingInterval_fast
		
	});	
}

module.exports = DucoMasterModbusDevice;

```

yaacov, 2023-12-21:
https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Promise/all

can you collect all reads, and then close after the last one is finished?

tjarb, 2023-12-21:
Hi @yaacov ,

thank you for your reply. I tried that, but returned an error that the body could not be read.

I'll try it again later today as suggested in the reference.

best regards,

tjarb, 2023-12-21:
Hi @yaacov ,

Bingo. I just past 100 reads.

```
const  ModbusRTU = require("modbus-serial");
let i=0;

self.pollingInterval_fast = self.homey.setInterval(() => {  
			
			let client = new ModbusRTU();
			self.log("####### "+ i++ +" poll to connect socket "+self.getSetting('address') +" ################");			
				client.setTimeout( 1000 );
				
				client.connectTCP( "192.168.2.173" , { port: 502 }).then( function(){					
					client.setTimeout( 1000 );
					client.setID(1);				
					
					Promise.all([
						client.readHoldingRegisters(10, 1),								
						client.readHoldingRegisters(15, 2),
						client.readHoldingRegisters(19, 1),
						client.readInputRegisters (10, 3),
						client.readInputRegisters(13, 3),
						client.readInputRegisters(19, 1)	
					]).then((data) => {
							self.log(data);
							client.close();
					}).catch((err) => {
					
						self.log(err);
					})							
											
				})	//client.close(); here the Modbus client is closed again
				.catch(function(e){		
					self.log(e);
				});	
									
}, 2000);
```