yaacov/node-modbus-serial #541: Destroy function callback is never called

narttmk, 2024-01-16:
[image]
this._port.destroy function is received an Error instance, so in this case callback never to be called

yaacov, 2024-01-16:
Hi, thank you for the issue.

Q. does the `callback` get called ?

Can you investigate the reason for this issue, and suggest a fix ?

Ref:
https://github.com/yaacov/node-modbus-serial/blob/master/index.js#L684

narttmk, 2024-01-16:
Hi yaacov, 
A: callback does not get called. 
Because SerialPortStream extend stream.Duplex
Ref: 
https://github.com/serialport/node-serialport/blob/main/packages/stream/lib/index.ts#L70
https://nodejs.org/api/stream.html#class-streamduplex

So the arg push into destroy function should be array of Error. 
I suggest that we should move callback to next line below destroy
[image]

yaacov, 2024-01-16:
nice, can you make a pull request ?

narttmk, 2024-01-17:
Sure, I created a PR, please help me to review and merge it
https://github.com/yaacov/node-modbus-serial/pull/542

closed 2024-01-17 by narttmk