yaacov/node-modbus-serial #544: Modbus RTU - CRC delay

Ababuy78, 2024-02-05:
I am observing a 500ms delay between any read command (FC1, FC2, FC3 or FC4) and its respective CRC.

Is this behaviour expected? It seems odd.

[image]

yaacov, 2024-02-05:
hi, thank you for the issue, can you find more information and fix if needed ?

Ababuy78, 2024-02-05:
I could try, but I am not proficient at javascript so it would require an unnecessary large amount of time to immerse my self on this project to find the issue and correct it.
I would prefer if someone familiarized with the project to correct this issue. Is that ok?

yaacov, 2024-02-05:
Sure, I added a "help wanted" label, It will let people know that can look at this issue and comment if they have more information, or find a fix.

thank you again for noticing this issue and reporting it :green_heart:

teddy1565, 2024-04-09:
HI, I want check a question
Are you using modbus-serial self-host a ModbusRTU Slave?

Ababuy78, 2024-04-09:
Hello!
No, I was using an external emulator.
As I could not see any problem with the modbus-serial library I connected directly to a PLC and the problem was solved.
It seems that the issue was using the same machine to test everything with a serial port emulator in the middle.
I believe this issue can be closed.
Thank you!