yaacov/node-modbus-serial #549: Error "Buffer is not defined" during start Angular application

p-gatto, 2024-03-21:
Hi,
With Angula17, nodejs v. 20.10 and modbus-serial 8.0.16 I'm trying to carry out readings with the modbus protocol using 'modbus-serial' but I have the following problem when starting the application highlighted in the console:

Uncaught ReferenceError: Buffer is not defined
     at addBufferBitOp (buffer_bit.js:30:5)
     at node_modules/modbus-serial/index.js (index.js:20:1)
     at __require2 (chunk-FWOMF3V2.js?v=883fa305:41:50)
     at index.js:1194:33

No errors are reported within the `buffer_bit.js` file at design time.
Thanks in advance for clarifications.

Paolo

yaacov, 2024-03-21:
Hi, thank you for the issue,

I am adding a help wanted label, so if someone knows the answer they will know to help.

p-gatto, 2024-03-21:
Hi,
In my opinion the problem lies in not correctly recalling the library resources within the Angular component. Below is the code I wrote:

import { Injectable } from '@angular/core';

@Injectable({
  providedIn: 'root'
})
export class ModbusService {

  constructor() { }

  readHoldingRegisters() {
  
    // create an empty modbus client
    const ModbusRTU = require("modbus-serial");
    const client = new ModbusRTU();

    // open connection to a tcp line
    client.connectTCP("192.168.1.54", { port: 502 });
    client.setID(1);

    // read the values of 10 registers starting at address 0
    // on device number 1. and log the values to the console.
    setInterval(function() {

      console.log('Ciao mondo!!!');
      /* client.readHoldingRegisters(0, 10, function(err, data) {
        console.log(data.data);
      }); */

    }, 1000);

  }

When I call the "readHoldingRegisters" function I get the following error:

core.mjs:6531 
 ERROR Error: Dynamic require of "modbus-serial" is not supported
    at main.js:6:9
    at _ModbusService.readHoldingRegisters (modbus.service.ts:13:23)
    at _HomeComponent.readHR (home.component.ts:18:25)
    at HomeComponent_Template_button_click_12_listener (home.component.html:8:72)
    at executeListenerWithErrorHandling (core.mjs:25518:16)
    at wrapListenerIn_markDirtyAndPreventDefault (core.mjs:25552:22)
    at HTMLButtonElement.<anonymous> (platform-browser.mjs:749:112)
    at _ZoneDelegate.invokeTask (zone.js:403:31)
    at core.mjs:14511:55
    at AsyncStackTaggingZoneSpec.onInvokeTask (core.mjs:14511:36)
Mostra 1 altro fotogramm

Anche se ho lanciato il seguente comando "npm i --save-dev @types/node" e registrato in tsconfig.ts nei types "node".
Thanks in advance for clarifications.

Paolo

teddy1565, 2024-04-09:
in my understanding
Angular is a frontend develop framework

modbus-serial is a backend library

One of the most obvious possibility questions
Frontend not support Buffer, Frontend only have ArrayBuffer
it's the biggest difference between front-end and back-end execution environments.

p-gatto, 2024-04-11:
Hi Teddy, You're right, I tried to manage it from the back-end and now I can communicate with the slave. Thanks for feedback!

aaelephant, 2024-04-30:
in vue too

FlowerSeaSea, 2024-05-17:
> 在我的理解中，Angular 是一个前端开发框架
> 
> modbus-serial 是一个后端库
> 
> 最明显的可能性问题之一：前端不支持 Buffer，前端只有 ArrayBuffer，这是前端和后端执行环境之间最大的区别。

如何转为不同类型的数据，比如float，short, word等，我试图用data.buffer.readFloatBE()转化但是得到的结果不正确，请问该如何正确拿到对应类型的值

teddy1565, 2024-05-17:
> > 在我的理解中，Angular 是一个前端开发框架
> 
> > 
> 
> > modbus-serial 是一个后端库
> 
> > 
> 
> > 最明显的可能性问题之一：前端不支持 Buffer，前端只有 ArrayBuffer，这是前端和后端执行环境之间最大的区别。
> 
> 
> 
> 如何转为不同类型的数据，比如float，short, word等，我试图用data.buffer.readFloatBE()转化但是得到的结果不正确，请问该如何正确拿到对应类型的值

Show your code?