yaacov/node-modbus-serial #552: I expect to be able to return a string in the getMultipleInputRegisters method.

Davensi, 2024-04-16:
当然，以下是您可以在 GitHub 上提交的 issue 的英文版本：

Issue Description:Issue Description:

I found that returning a string in the getMultipleInputRegisters method seems not to be supported when using modbus-serial to create a slave.

Steps to Reproduce:

(List the steps to reproduce the issue)
Expected Behavior:

I expect to be able to return a string in the getMultipleInputRegisters method.

Actual Behavior:

In reality, an array containing the byte sequence of the string encoded in GBK format is returned.

Additional Information:

In the getMultipleInputRegisters method, I attempted to convert a string into a byte sequence encoded in GBK format and return it as an array. However, this does not seem to work.

Here is my code:

I found that returning a string in the getMultipleInputRegisters method seems not to be supported when using modbus-serial to create a slave.

Steps to Reproduce:

(List the steps to reproduce the issue)
Expected Behavior:

I expect to be able to return a string in the getMultipleInputRegisters method.

Actual Behavior:

In reality, an array containing the byte sequence of the string encoded in GBK format is returned.

Additional Information:

In the getMultipleInputRegisters method, I attempted to convert a string into a byte sequence encoded in GBK format and return it as an array. However, this does not seem to work.

Here is my code:
```
const iconv = require('iconv-lite');
// Omit some code
getMultipleInputRegisters: function (startAddr, length) {
    console.log('[ getMultipleInputRegisters ] >', startAddr, length, Date.now())
    const values = [];
    if (startAddr >= 5000 || startAddr <= 6000) {
        const str = "        开始时间              结束时间        落纱耗时   落前断头 启动断头 运行断头 留头率  纺纱长度"; // Replace with the string you want to return
        const gbkEncodedBuffer = iconv.encode(str, 'gbk');

        // Write the GBK encoded byte sequence into the values array
        for (let i = 0; i < gbkEncodedBuffer.length; i++) {
            values.push(gbkEncodedBuffer[i]);
        }
        console.log('[ values ] >', values,values.length)
        return values
    }
    for (let i = 0; i < length; i++) {
        values[i] = inputRegisters[startAddr + i];
    }
    return values
}
```

yaacov, 2024-04-16:
Hi,
Thank you for your issue.

I added a label for help, maybe others have the same issue.

My suggestion is to check that you get the buffer, check the content, and length, check character size ( the device might encode a char in one byte, two bytes, etc ...), check endianness ( the device might use big or little  ), see:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/debug.js#L36

Davensi, 2024-04-16:
I've tried some other approaches, such as writing a fixed-length string to return, but none of them seem to work as expected.

Davensi, 2024-04-16:
Here is my code. I'm hoping that when the getMultipleInputRegisters method is triggered, it returns a string, which would facilitate displaying content on my MCGS screen. However, it seems that this approach is not viable.
```
// create an empty modbus client
const ModbusRTU = require("modbus-serial");
const iconv = require('iconv-lite');
// 保持寄存器
const holdingRegisters = { 4000: 31, 4001: 32, 4002: 25, 4003: 26, 4004: 27, 4100: 0, 4101: 10, 4102: 1, 4103: 0 };
// 线圈 0-off 1-on 
const coils = {
    0: 0,
    1: 0,
    2: 0,
};
// 读输入寄存器
const inputRegisters = {};
// 离散输入 开关量输入
const discreteInputs = {};
const vector = {
    // 获取单个输入寄存器
    getInputRegister: function (addr) {
        console.log('[ getInputRegister ] >', addr)
        return inputRegisters[addr];
    },
    // 获取多个寄存器
    // 1-That's accurate ？
    getMultipleInputRegisters: function (startAddr, length) {
        console.log('[ 获取多个输入寄存器getMultipleInputRegisters ] >', startAddr, length, Date.now())
        const values = [];
        if (startAddr >= 5000 || startAddr <= 6000) {
            const str = "        开始时间              结束时间        落纱耗时   落前断头 启动断头 运行断头 留头率  纺纱长度"; // 替换成你想要返回的字符串
            const gbkEncodedBuffer = iconv.encode(str, 'gbk');

            // 将 GBK 编码的字节序列写入到 values 数组中
            for (let i = 0; i < gbkEncodedBuffer.length; i++) {
                values.push(gbkEncodedBuffer[i]);
            }
            console.log('[ values ] >', values,values.length)
            return values
        }
        for (let i = 0; i < length; i++) {
            values[i] = inputRegisters[startAddr + i];
        }
        return values
    },
    // 获取分离输入
    getDiscreteInput: function (addr) {
        console.log('[ getDiscreteInput ] >', addr)
        return discreteInputs[addr];
    },
    // 获取保持寄存器
    getHoldingRegister: function (addr) {
        console.log('[ getHoldingRegister ] >', addr)
        return holdingRegisters[addr];
    },
    // 设置单个寄存器
    setRegister: function (addr, value) {
        // 设置报表ID时
        if (addr === 4102) {
            console.log('[ 设置报表ID ] >', value)
        }
        holdingRegisters[addr] = value;
        return;
    },
    // 获取多个保持寄存器
    getMultipleHoldingRegisters: async function (startAddr, length, unitID, callback) {
        const values = [];
        for (let i = 0; i < length; i++) {
            values[i] = holdingRegisters[startAddr + i];
        }
        callback('', [9, 8, 62, 134])
       
    },
    // 获取一个线圈
    getCoil: async function (addr, id) {
        // cd(coils[addr])
        return coils[addr];
    },
    // 设置一个线圈
    setCoil: function (addr, value) {
        coils[addr] = value;
        return coils[addr];
    },
    // 读取设备号
    readDeviceIdentification: function () {
        console.log('[ readDeviceIdentification ] >')
        return {
            0x00: "MyVendorName",
            0x01: "MyProductCode",
            0x02: "MyMajorMinorRevision",
            0x05: "MyModelName",
            0x97: "MyExtendedObject1",
            0xab: "MyExtendedObject2"
        };
    }
};
const serverTCP = new ModbusRTU.ServerSerial(vector,
    {
        port: "COM6",
        baudRate: 115200,
        debug: true,
        unitID: 1,
        interval: 10,
    }
);

```

yaacov, 2024-04-16:
https://github.com/yaacov/node-modbus-serial/blob/master/servers/serverserial.js#L164

are you trying to implement `readInputRegisters`  ?
Ref: 
https://github.com/yaacov/node-modbus-serial/blob/master/servers/serverserial.js#L164

yaacov, 2024-04-16:
you can try and look what happen in https://github.com/yaacov/node-modbus-serial/blob/master/servers/servertcp_handler.js#L416

Davensi, 2024-04-16:
Thank you. I found a solution. I should initialize it in my variable in advance. Actually, it's enough to send the corresponding GBK code of the MCGS screen.

```
const setTable = (reportId) => {
    if (reportId === 0) {
        return []; // 如果 reportId 为 0，则返回空数组
    }
    const str = titleMap[reportId]; // 假设 titleMap 是在其他地方定义的对象
    const gbkBuffer = iconv.encode(str, 'gbk'); // 使用 iconv 将字符串转换为 GBK 编码的缓冲区
    const registers = []; // 用于保存 16 位寄存器值的数组
    let j = 0
    for (let i = 0; i < gbkBuffer.length; i += 2) {
        const value = gbkBuffer.readUInt16LE(i); // 读取 16 位小端序值
        registers.push(value);
        // inputRegisters[j + 5000] = value
        j++
    }
    // console.log('[ inputRegisters ] >', inputRegisters)
    // return Buffer.from(registers); 
    return registers;
}
```

closed 2024-04-16 by Davensi