yaacov/node-modbus-serial #554: ServerTCP getMultipleHoldingRegisters and getMultipleInputRegisters don't work with Promises

ensehic, 2024-05-13:
I'm having issues implementing async `getMultipleInputRegisters` and `getMultipleHoldingRegisters`. See code below.

```
import { ServerTCP } from 'modbus-serial';

const vector = {
    getMultipleInputRegisters: (addr: number, length: number, unitID: number) => {
        return new Promise((resolve) => {
            resolve([1, 2]);
        });
    },
    getMultipleHoldingRegisters: (addr: number, length: number, unitID: number) => {
        return new Promise((resolve) => {
            resolve([3, 4]);
        });
    }
};

const serverTCP = new ServerTCP(vector, { host: '0.0.0.0', port: 502, unitID: 1 });
serverTCP.on('error', console.log);
```

When I attempt to read multiple input registers or holding registers (trying to read, say, address 10 and length 2), I receive this error on the client side:
```
{
    status: 'rejected',
    reason: Error: Modbus exception 4: Slave device failure (device reports internal error)
        at ModbusRTU._onReceive (...)
        at TcpPort.emit (node:events:518:28)
        at TcpPort.emit (node:domain:488:12)
        at Socket.<anonymous> (...)
        at Socket.emit (node:events:518:28)
        at Socket.emit (node:domain:488:12)
        at addChunk (node:internal/streams/readable:559:12)
        at readableAddChunkPushByteMode (node:internal/streams/readable:510:3)
        at Socket.Readable.push (node:internal/streams/readable:390:5)
        at TCP.onStreamRead (node:internal/stream_base_commons:190:23) {
      modbusCode: 4
    }
}
```

The error message is rather generic, so I haven't been able to glean anything helpful from it. I have verified that both these methods work when used synchronously. Also all other methods (`setCoil`, `getInputRegister` etc.) work fine with Promises. I imagine I'm doing something wrong here, but can't for the life of me figure out what it is. Any help would be appreciated. 🙂

ensehic, 2024-05-14:
Turns out I was indeed doing something wrong.

paloky, 2024-11-12:
Hi ensehic.
I have also had problems with the function  'getMultipleHoldingRegisters'.

I solved with this:

1.- Insert "async" to function.
2.- Use "await" to obtain the values you want return.
3.- Use "callback" to return values.

```
//*****************************************************************
// Event Function 3 (Read Multiple Holding Register)
//*****************************************************************
getMultipleHoldingRegisters: async function (startAddr, length, unitID, callback) {
    console.log(`Reading Multiple Holding Registers StartAddr[${startAddr}] with Length[${length}]`);

    // Get the registers values from DB
    let valores = await Get_DB_Registers(startAddr, length);

    // Prepare an Array to return values.
    const values = [];
    for (let i = 0; i < length; i++) {
        if (isNaN(parseInt(valores[i]))) {
            values[i] = 0;
        } else {
            values[i] = parseInt(valores[i]);
        }
    }

    // Return values using 'callback'
    console.log("Returned Values ", values);
    callback(null, values);
},
```

This work for me.
Best Regards.