yaacov/node-modbus-serial #561: modbus - baudRate 115200 - does not work

robert-beecloud, 2024-08-12:
Hi,
nod-modbus-serial works only when baudRate is 9600.
When I change it to 19200 or 115200 - it does not work :(

It is very important for my project. Serial port is working with my NFC card reader on 115200 and I need to connect to my diagnostic tool over modbus with the speed higher then default: 9600.

My simple code below. I tested it also with serialPort object and without it

var {SerialPort} = require("serialport");
var serialPort = new SerialPort({ path: "/dev/ttymxc2", baudRate: 115200, autoOpen: false});

// create an empty modbus client
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU(serialPort);

// open connection to a serial port
//client.connectRTUBuffered('/dev/ttymxc2', { baudRate: 9600}, run);

client.open(run);

function run() {
    client.setID(1);
    if (client.isOpen)
      client.readInputRegisters(0, 10, function(err, data) {
        if (err) {
                console.log(err);
        }
        else {
                console.log(data.data);
        }
    });
}
setInterval(run, 1000);

Thanks for any help!

yaacov, 2024-08-12:
Hi, thank you for the issue, I added a help wanted flag so people that may help will know to look here

robert-beecloud, 2024-08-14:
Hi,
I have tested my configuration and I solved it!
node-mobus-serial is not a problem - it works perfect ;)
Problem was with USB-RS485 converter!
Everybody should know that working with USB-RS485 converter for 5 USD is not a good idea.
Thanks.