yaacov/node-modbus-serial #571: Can a Modbus TCP Slave actively send a frame to a Modbus TCP Master?

Oniokey, 2024-12-14:
I looked through various examples and documentation and found that a slave cannot actively send a frame of data to the master, but this can be useful in certain scenarios, such as when an IoT server is acting as a slave, one might want to be able to send down some configurations from the cloud to a device, which would require the slave to be able to actively send data to the master.

So, how can I send data from slave to master?

Stormpass, 2025-01-02:
When the slave station sends data to the master station,  i think it just calling the write register API?

Oniokey, 2025-01-06:
Sorry, but I didn't get it. When I start a slave station, I use this
```ts 
import { ServerTCP， IServiceVector } from "modbus-serial"

const vector: IServiceVector = { /* some functions here */ }

const client = new ServerTCP(vector, {host: "0.0.0.0", port: 8502, debug: true, unitID: 1});
```
but `client` did not have the write register API

Xsandor, 2025-04-10:
You need to read about Modbus and you will see that it's a master/slave protocol. The only way to exchange information between a  slave and a master is by having the master send a request to the slave.

The whole protocol is built around the concept of request => response. 

This is not a limitation in this library, it's a limitation in the protocol.

When slaves start sending data without it being requested they are no longer Modbus devices, since they aren't respecting the specification.