yaacov/node-modbus-serial #572: getMultipleHoldingRegisters() in Modbus-TCP Server causes a failure when returning a Promise

romor, 2025-01-09:
According the TypeScript definitions, `IServiceVector::getMultipleHoldingRegisters()` support returning `Promise<number[]>`.

However, when this is used, the return length check in `servertcp_handler.js::300` will always fail. The corresponding code part is:

            const values = vector.getMultipleHoldingRegisters(address, length, unitID);
            if (values.length === length) {

As `values` is `Promise{[...]}` its length check fails and this results in a Modbus exception sent to client instead of the provided read result.

When `getMultipleHoldingRegisters()` returns `number[]` directly, without a Promise, everything works as expected.

modbus2mqtt, 2025-05-07:
I have the same problem
Workaround:
use callback function by wrapping the promise
```
  queueMultipleRegister( addr: number, length: number, unitID: number, cb: FCallbackVal<number[]>): void{
    yourPromise(
      addr,
      length,
      unitID
    ).then((value)=>{
        cb(null,value)
    }).catch(e=>{cb(e,[])})
  }
``` 

However, it would be nicer if it works. :-)

volkmarnissen, 2025-05-22:
It's not only HoldingRegister which doesn't work with Promise, it's also Coils
Additionally  the Promise.then method for  writeRegister will not be called after successful execution

It's very hard to find these kind of issues.
So, they should either work properly or marked as "not working" in the documentation.

I replaced all calls with promisses to calls with callback. This is more reliable.