yaacov/node-modbus-serial #589: `no-serial-port` actually requires `serialport`

IlCallo, 2025-04-18:
I'm trying to use latest `8.0.20-no-serial-port` version in an ESM project
It does indeed require `serialport` in multiple occasions, making the whole package less portable (I'm bundling a Node app with Bun compilation)

These are the lines that force a dependency on `serialport`:

https://github.com/yaacov/node-modbus-serial/blob/94c1a3ebc7f3ae275932021ee7bda9917c7327e6/index.js#L21

https://github.com/yaacov/node-modbus-serial/blob/94c1a3ebc7f3ae275932021ee7bda9917c7327e6/index.js#L1240

https://github.com/yaacov/node-modbus-serial/blob/94c1a3ebc7f3ae275932021ee7bda9917c7327e6/index.js#L1256-L1258

https://github.com/yaacov/node-modbus-serial/blob/94c1a3ebc7f3ae275932021ee7bda9917c7327e6/index.js#L1264-L1265

Dunno if something changed recently or the package just isn't optimized to be executed within ESM projects

yaacov, 2025-04-18:
hi, thank you for noticing that,
can you make a pull request removing the hard dependencies on serial port ?
for example, ignoring import error, and check for the lib before calling code that require it, and quietlyfall back if needed?

yaacov, 2025-04-18:
p.s. if you can think of a test to know the core lib will not work without serial, please add a test

yaacov, 2025-04-19:
https://github.com/yaacov/node-modbus-serial/pull/590

yaacov, 2025-04-19:
https://www.npmjs.com/package/modbus-serial/v/8.0.21-no-serial-port 🎉 

can you test 8.0.21 ^^

IlCallo, 2025-04-20:
Thanks! 

I'll test it out Tuesday 👍🏻

IlCallo, 2025-04-22:
Tested this, it does work with PNPM build

It does not work with Bun compilation tho
It seems like Bun tries to execute the Bun require anyway on its own, thus bypassing the try/catch you added

I don't know enough to tell you how and if this can be fixed unluckily

yaacov, 2025-04-22:
Thank you for testing,
can you atach here a script or a test, I can use the test if a fix work for bun ?
e.g.
install bun, run A folowed by B, see expected result Y

IlCallo, 2025-04-22:
I'm digging a bit deeper

A possible problem could be the one mentioned here: https://bun.sh/docs/bundler/executables#unsupported-cli-arguments

> always outputs a binary executable. Internally, it's almost esm.

But I think the **actual** problem we're hitting is [this one](https://github.com/oven-sh/bun/issues/14509):

> For context, bun build tries to bundle every import, assuming they’ll be needed. In a normal runtime, this import wouldn’t be encountered, but during the build process, Bun attempts to bundle it and encounters this issue.

I tried just removing all problematic imports that would cause the problem, but something went from (dunno if due to that or due to Bun itself) since that broke the app

If I build the app with pnpm and run it as a normal Node built project, not as a standalone executable, the new version you released works like a charm

---

To test this, create a Bun project, then create a `src/index.ts` file and add this content
```ts
import ModbusRTU from "modbus-serial";

const client = new ModbusRTU.default();
```

Then run `bun build --compile --target=bun-windows-x64 --minify --sourcemap src/index.ts --outfile dist/test`

I haven't tested the procedure, but should work as the problem arise as soon as you reference ModbusRTU value (not the type, which is stripped away before compiling) into the code

IlCallo, 2025-04-22:
My temporary workaround: using `bun patch` to replace all `require('serialport')` with `{ SerialPort: {} }`

To solve the problem in the long run, the codebase should probably be restructured to isolate in a single file all pieces that require `serialport`, then you'd probably setup your build process to generate due different bundles:
- one which installs `serialport` and keeps the `require('serialport')`
- one that automatically excludes the file with all `require('serialport')` usages, or that replaces all `require('serialport')` with `{ SerialPort: {} }`

---

## My workaround

Pin `modbus-serial` dependency, or it will evenually go out of sync

Add in `package.json`
```
"patchedDependencies": {
    "modbus-serial@8.0.21-no-serial-port": "patches/modbus-serial@8.0.21-no-serial-port.patch"
  }
```

Add bun patch file

File name `patches/modbus-serial@8.0.21-no-serial-port.patch`
```
diff --git a/apis/connection.js b/apis/connection.js
index 7f2acd1dae5222ee9338b2143c9d00b71e5e9420..778899e6975e55d5d33ee59b5e29036f8be2f150 100644
--- a/apis/connection.js
+++ b/apis/connection.js
@@ -79,7 +79,7 @@ const addConnctionAPI = function(Modbus) {
         options.platformOptions = { vmin: MIN_MODBUSRTU_FRAMESZ, vtime: 0 };
 
         // create the SerialPort
-        const SerialPort = require("serialport").SerialPort;
+        const SerialPort = { SerialPort: {} }.SerialPort;
         this._port = new SerialPort(Object.assign({}, { path }, options));
 
         // open and call next
diff --git a/index.js b/index.js
index 2ecb02e4c4ffa4061cc3102d4bcc19a19defe2d8..816510c734f73f9865ad051613d191e1fd1baa92 100644
--- a/index.js
+++ b/index.js
@@ -1248,7 +1248,7 @@ require("./apis/worker")(ModbusRTU);
 module.exports = ModbusRTU;
 
 module.exports.getPorts = function getPorts() {
-    const { SerialPort } = require("serialport");
+    const { SerialPort } = { SerialPort: {} };
 
     return SerialPort.list();
 };
diff --git a/ports/asciiport.js b/ports/asciiport.js
index aecee5adf1a99234ebc9940c2b8b3e5e2ba9e8af..76b82f63c67500c6d3897b070b52ee7791f1a8f1 100644
--- a/ports/asciiport.js
+++ b/ports/asciiport.js
@@ -3,7 +3,7 @@
 
 const events = require("events");
 const EventEmitter = events.EventEmitter || events;
-const SerialPort = require("serialport").SerialPort;
+const SerialPort = { SerialPort: {} }.SerialPort;
 const modbusSerialDebug = require("debug")("modbus-serial");
 
 const crc16 = require("../utils/crc16");
diff --git a/ports/rtubufferedport.js b/ports/rtubufferedport.js
index 6260947d87cb879a248d9c9615fe2bbab8992056..99a2d9a207332e93d3fe7e7f6c65cce3bd98c3d7 100644
--- a/ports/rtubufferedport.js
+++ b/ports/rtubufferedport.js
@@ -1,7 +1,7 @@
 "use strict";
 const events = require("events");
 const EventEmitter = events.EventEmitter || events;
-const SerialPort = require("serialport").SerialPort;
+const SerialPort = { SerialPort: {} }.SerialPort;
 const modbusSerialDebug = require("debug")("modbus-serial");
 
 /* TODO: const should be set once, maybe */
diff --git a/servers/serverserial.js b/servers/serverserial.js
index c30fb983cafe104eef31a000ae5dce174b4b5a8d..100245bde0b2953768dddd8db1e091879537ef0f 100644
--- a/servers/serverserial.js
+++ b/servers/serverserial.js
@@ -17,7 +17,7 @@
 const events = require("events");
 const EventEmitter = events.EventEmitter || events;
 const modbusSerialDebug = require("debug")("modbus-serial");
-const { SerialPort } = require("serialport");
+const { SerialPort } = { SerialPort: {} };
 const ServerSerialPipeHandler = require("./serverserial_pipe_handler");
 
 const PORT = "/dev/tty";
```

Run `bun install` to update the `bun.lock`

IlCallo, 2025-04-22:
Yeah, that would be centralizing all `require` into a file, so that shoud be good

gil-obradors, 2025-09-04:
Hi! I'm testing modbus-serial with a Trigger.dev server because I need it for remote TCP polling. By default, this setup doesn’t work well in that kind of infrastructure.

Trigger.dev spins up small Docker containers for scheduled tasks, but there’s a workaround: you can install the package outside the build process. Configuring trigger.config.ts like this works perfectly.

```
import { defineConfig } from "@trigger.dev/sdk/v3";

export default defineConfig({
  ...
  build: { external: ["modbus-serial", "serialport"] },
});
```

I have no idea how it could be improved, but it's still very good work!